foostan/crkbd #309: Can't connect keyboard on Linux

blgram, 2025-08-08:
Hello,

This mightnot be technical question. If it's dump, please disregard this post. 

I recently purchased a crkbd/rev4.1 standard, whereas when it's connected to my windows pc, I am able to use vial web to manage the keybaord, but I can't do it in Linux, It stucks at "connecting" 

I use Linux as my primary device, so I would like to be able to change keys on go to make it fit my test.

If you have any idea what could be the reason, I will check and share logs if it's neccessary. 

Regards,

miguelbaldi, 2025-08-08:
Vial documented exactly this:

https://get.vial.today/manual/linux-udev.html

blgram, 2025-08-08:
Thank yo so much, this fixed the issue. 

I am not so good at LInux yet, I knew it was a udev issue, but didn't know how to resolve it. 

I wish I looked further in vial's website, I did look but it seemed I missed this. 

thanks again.

closed 2025-08-08 by blgram

miguelbaldi, 2025-08-08:
> Thank yo so much, this fixed the issue.
> 
> I am not so good at LInux yet, I knew it was a udev issue, but didn't know how to resolve it.
> 
> I wish I looked further in vial's website, I did look but it seemed I missed this.
> 
> thanks again.

You're welcome!