foostan/crkbd #311: Unclear instructions for cases and switchplates

YourPsychiatrist, 2025-09-06:
Hello!

First, thanks a lot for providing the output of your work on this keyboard for free.

I have had the keyboard assembled by JLCPCB, tested it for functionality (checking for shorts, flashed the mcu, checked if it does keyboard stuff), and so far it all looks good. 

Now I'll assemble the rest of it, but I keep wondering:

1. The case you provide as step file is completely solid on the bottom, meaning that the LEDs wouldn't be able to shine through. Is that intended?
1.  In the [build instructions](https://github.com/foostan/crkbd/blob/main/docs/corne-chocolate/v4/buildguide_en.md), you mention that a switch plate is required. My question is, what am I supposed to do with [these plates](https://github.com/foostan/crkbd/tree/main/plates/3x6)? 
    - These look again like they are supposed to be ordered from jlcpb, but they aren't actual PCBs. Can these not be 3D-printed?
    - If i were to copy the edge cuts lines to freecad and extrude  the surface by 1.6mm for 3d printing, which one would i need for the corne choc 4.1? `plate-top-alps`? `plate-top-alps-ex?`?
    - The build instructions only show a top plate, but the folder contains a `plate-bottom`. What is that for?

If it turns out that I'll have to create some models on my own, I'd be happy to share them through a PR if you want.

YourPsychiatrist, 2025-09-07:
> There are no under side LEDs from v4.

- [schematic version](https://github.com/foostan/crkbd/blob/main/pcbs/corne-chocolate/hotswap/corne-chocolate.kicad_sch#L10)
- [leds](https://github.com/foostan/crkbd/blob/main/pcbs/common/left.kicad_sch#L1899)

> Use can use https://github.com/foostan/crkbd/tree/main/plates/3x6/pcb files to order like other PCB file.

i understand that i can use that. My question is if there is a reason this isn't available as STP file for 3d printing. Would that not be sturdy enough?

> It's for sandwitch style

ah cool, thanks

foostan, 2025-09-07:
I'm not sure if the strength is sufficient. I simply assumed that for something as simple as a flat plate, the SVG or KiCAD data for FR4 would be enough.

anannaranj, 2025-11-08:
I think v4 build guide isn't as well documented as the v3 or v2, right?